spacehuhn/wifi_ducky #1: Arduino Pro mini?

H1ghTech, 2017-03-24:
Just wondering if this could work if you add a usb with d+ d- + -

tobozo, 2017-03-24:
Depending on the chosen model (pro mini 3.3v or 5v) I wonder if you'll need a logic level converter between the arduino and the ESP or between the arduino and the USB.

H1ghTech, 2017-03-24:
Ok I got a 5v one but I have a level shifter

tobozo, 2017-03-24:
I'll try later tonight, I have both 5v and 3.3v model, I'm very interested by the results of your tests

H1ghTech, 2017-03-24:
And how do i wire the usb jack?😂

spacehuhn, 2017-03-24:
you need an ATmega32u4 the Pro mini doesn't use that microcontroller.

spacehuhn, 2017-03-24:
But you could use a Arduino Pro micro. You'll need the 3.3V version yes, or use a voltage regulator.

H1ghTech, 2017-03-24:
What is the difference between the two microcontrollers?

spacehuhn, 2017-03-24:
one can talk USB the other not.

H1ghTech, 2017-03-24:
Aren't there any library's for usb ?

tobozo, 2017-03-24:
[They're not that expensive](https://www.aliexpress.com/item/Mini-Leonardo-Pro-Micro-ATmega32U4-5V-16MHz-Module-For-Arduino-Best-Quality/32284746884.html)

H1ghTech, 2017-03-24:
I didn't was that they are expensive lol

tobozo, 2017-03-24:
edited my comment, some of it got lost in transit

H1ghTech, 2017-03-24:
Lol

tobozo, 2017-03-24:
Just stumbled upon [this](https://github.com/cnlohr/espusb) and it looks like the ESP8266 can act as a low speed USB if you can bother to lose ESP8266AsyncWebServer and use SDK 1.5.x or earlier

spacehuhn, 2017-03-24:
haha @tobozo you haven't read the readme, right? Because I linked exectly that project. 😄

tobozo, 2017-03-24:
Your readme says "The Latest SDK" while espusb's readme says "You cannot use SDKs newer than 1.5.X" now I'm confused

tobozo, 2017-03-24:
Also why are you using an arduino when espusb can have the ESP to act as a host ?

spacehuhn, 2017-03-24:
### How it works

[...]

Although the ESP8266 is awesome, it doesn't have native USB, which means it can't act as a keyboard :(  
([cnlohr](https://github.com/cnlohr) made a cool project on this and added a USB stack himself: https://github.com/cnlohr/espusb. The problem with that is, that it isn't compatible with the current SDK version, also I wanted to use Arduino to make it more user friendly.)  

Here comes the ATmega32u4 into play!  
[...]

tobozo, 2017-03-24:
Ok danke stupid me lol

So in order to use this librarty one would need to use two ESPs, one for the logic+wifi, the other to act as a USB host, and that would be:

- An IDE mess
- Not user friendly
- Interferences
- Pointless

thank you sir!

closed 2017-03-25 by spacehuhn