spacehuhn/wifi_ducky #3: DigiSpark Support

spacehuhn, 2017-03-25:
Ok so I tried to port the atmega part onto the digispark, but there are a few problems:

**No Serial Port**
That problem can be solved by using software serial., but you can't use that together with the keyboard library.
Luckily someone wrote a library which solves that problem: https://github.com/J-Rios/Digispark_SoftSerial-INT0
But the serial port is still buggy and I couldn't get it to work properly.

**Lack of Keyboard Keys**
There aren't all keys supported in the digispark keyboard library. The Arduino keyboard library is way better! I'm sure you could still send every key, but to find what byte is for which key means a lot of try and error.
Not impossible, though very annoying.
And then you still have the problem with the different keyboard layouts for each country.

**Not enough Memory**
Getting the ducky script interpreter on the attiny85 isn't that easy, it's to big!
Maybe with clever code design it should be possible.
An other approach here is to let the ESP8266 interpret the Ducky Script and just send the plain keys to the DigiSpark. However... that's a lot of work.

I haven't given up on this yet, however any help is much appreciated!

lyudah, 2017-04-26:
hi, I wrote some scripts and i was wondering what other people came up with. Have you found a sort of library yet like there is for the rubber ducky?

knows, 2017-04-27:
hmmmmmm. why i'm @ed?

lyudah, 2017-04-27:
sorry, wasn't mean that way... it was a reference/ inside joke... your tag seemed to fit

TheTerminater01, 2017-05-18:
I would love to colab with your on this enhancement ill keep you updated with what I find out.