spacehuhn/wifi_ducky #12: Working 5v setup

tobozo, 2017-04-07:
Not really an issue but maybe this is worth updating the readme to menion this; here's my working configuration:

- Wemos Mini D1
- Arduino Micro 5v Edition

The 5v pin on the Wemos is plugged to the VCC on the Arduino. I'm not sure if this is a valid use case as the result is much bigger and probably more easily exposed, but it seems to work just fine.

Pictures of the assembly:

[image]
[image]

manchmal2, 2017-06-27:
Hey man
So this means that you can hook up a arduino that runs on 5v to a nodemcu that take 3v?
Can you explain a little more maybe?

manchmal2, 2017-06-27:
Sorry i got that messed up
So the wemos 3v can be connected to the nodemcu 5v? If yes can you explain it a little more im still new to this

br0k3nilluzion, 2017-06-28:
Manchmal2  the VCC on the Pro Micro is connected to the 5v on the D1 mini board.. look at the 5 next to the red wire.

manchmal2, 2017-06-28:
Sorry my phone only loaded the first picture...

manchmal2, 2017-06-28:
So to be completley sure i dont need any adapter i can just straight up connect them this way? Sorry for being a noob...

tobozo, 2017-06-28:
That's a 5v version of the arduino pro micro, if using the 3.3v version better power it from the 3v pin.

Wemos D1 is 5v tolerant so you don't necessarily need a level shifter.