spacehuhn/wifi_ducky #13: [Enhanchement]  "Is-anyone-around?" function

whid-injector, 2017-04-14:
I believe this function might be useful in wifi ducky!

Add, to the Atmega sketch, a function that will constantly check status changes of CAPSLOCK led and keep updated a red/green flag within each webpage in the ESP HTTP Server.
In order to detect if the victim is using the keyboard and potentially become aware of an ongoing attack.
In this way, the attacker (connected through wifi) will be constantly aware if someone is around the victim machine. This might increase the success rate and reduce the suspiciousness among victims.

P.S. Eventually it could even be deployed with a preset automatic payload, that will run as soon as wifi ducky will not detect any victim's activity.

Something like this... (WIP)
```
// within Arudino Sketch
boolean capsIsON(){
  if (BootKeyboard.getLeds() & LED_CAPS_LOCK){
    return true;
  }
  else{
    return false;
  }
}

void pressCAPSLOCK(){
  Keyboard.press(KEY_CAPS_LOCK);
  delay(100);
  Keyboard.release(KEY_CAPS_LOCK);
}

void loop() {  
  while (Serial1.available()) {
    
    if (capsIsON()){
      pressCAPSLOCK();
      while (!capsIsON()){ 
         // update html button to GREEN. Attack can go on!
         Serial1.print("GREEN*");
      }
      // someone changed CAPSLOCK status! Be Careful! Someone is around!
    }
    else  {
      pressCAPSLOCK();
      while (capsIsON()){
         // update html button to GREEN. Attack can go on!
         Serial1.print("RED*");
      }
      // someone changed CAPSLOCK status! Be Careful! Someone is around!
    }

//Sketch Continues...
}
```

```
// ESP Sketch
server.on("/", [](){
    String buttonStatus = "RED-original";
    buttonStatus = Serial.readStringUntil('*');
    server.send(200, "text/html", "<style>body {background-color: #000000;}.moveimage{position: relative;left: 55px;}</style><html><body><meta http-equiv=\"refresh\" content=\"3\" /><h1 style=\"color: #00ff00;\">WHID Injector</h1><p><span style=\"color: #00ff00;\">WiFi HID Injector for Fun & Profit. The Button is: "+String(buttonStatus)+"</span></p><p><span style=\"color: #00ff00;\">-----------------------------------------------</span></p><a style=\"color: #00ff00;\" href=\"/uploadpayload\">Upload Payload</a></html><br>-<br><a style=\"color: #00ff00;\" href=\"/listpayloads\">Choose Payload</a><br>-<br><a style=\"color: #00ff00;\" href=\"/format\">Format File System</a></html><p><span style=\"color: #00ff00;\">-----------------------------------------------</span></p>");
  });
```

spacehuhn, 2017-04-14:
Have you tested this?
Because wouldn't you only change the caps lock on the Arduino HID but not on the other keyboard?

whid-injector, 2017-04-14:
Didn't have time yet to finish the sketch, but indeed it should work.
Forgot to mention that it requires 
```
#include <HID-Project.h>
#include <HID-Settings.h>
```
Source: https://github.com/NicoHood/HID/blob/master/examples/Keyboard/KeyboardLed/KeyboardLed.ino

The idea is:  once inserted the USB dongle, the 32u4 would immediately change the CAPSLOCK state and then keep listening if someone is changing it back.
Of course this is just an example of how to detect victim's around... ;)