spacehuhn/wifi_ducky #15: [Feature Request] OTA Update

tobozo, 2017-04-18:
I found a practical use to this toy: Retropie shortcuts! No more keyboard required, now I can send pause/play, insert-coin, exit, and other emulator related shortkeys from the phone.

Problem: version 1.1 requires to reflash the ESP, but reflashing ESP is impossible when rx/tx are soldered. 

Since I'll have to cut the wires anyway (damnit why did I remove [the switch](https://github.com/spacehuhn/wifi_ducky/issues/12) ?), I'll add the OTA capability to the code before reflashing.

My question is: if I want to be able to start the WifiDucky OTA update state from the web controls, all I need is to add a HTML button, a web handler and implement the OTA logic, right ?

spacehuhn, 2017-04-18:
https://github.com/spacehuhn/wifi_ducky#update-esp8266-over-the-webinterface

Go to 192.168.4.1/update and upload the new .bin file.

tobozo, 2017-04-18:
okay thanks that did it \o/

closed 2017-04-18 by tobozo