spacehuhn/wifi_ducky #16: Ducky script parsing issue

whid-injector, 2017-04-27:
Steps to reproduce:
- create the following script and execute it:
```
GUI r
DELAY 500
STRING powershell Start-Process cmd -Verb runAs
```
It will stop typing at **powershell Start-**

[image]

spacehuhn, 2017-04-27:
Works perfekt for me (also on the WHID).

whid-injector, 2017-04-28:
https://youtu.be/BMy_yeNlIBw

Here full PoC of the issue.
My default layout is the standard US (just to be precise).

whid-injector, 2017-04-28:
As counter proof that with WHID's sw it works... 
https://youtu.be/IaAmQoOjrNE

BlueArduino20, 2017-04-29:
This problem only happens when we put a capital letter after a "-". I've got the same problem.

spacehuhn, 2017-04-29:
Then it should work if you put a delay between it.

whid-injector, 2017-05-02:
I tried with the delay, but no work.

@BlueArduino20 which baudrate are you using? 57200bps? can you post a video of your case?

spacehuhn, 2017-05-02:
pretty sure this is related to your keyboard layout.

whid-injector, 2017-05-02:
I believe there was some issue regarding the CR/LF differences between Win and Linux. 
I solved, by writing by hands the full payload instead of copy-pasting it.
P.S. The layout was not a problem, since I am always testing with US.

BlueArduino20, 2017-05-02:
I've solved it by modifying the Keyboard library. Yes, we can close the issue.

whid-injector, 2017-05-02:
@BlueArduino20  can you write more details about your mods?
Do you have a fork that I can look at?

BlueArduino20, 2017-05-02:
I've just used this: https://github.com/Nurrl/LocaleKeyboard.js
You can choose the locale what you want.

whid-injector, 2017-05-02:
Cool. Closing issue.

closed 2017-05-02 by whid-injector