spacehuhn/wifi_ducky #17: Compiling error

pyto00, 2017-05-03:
When I try to compile the Arduino_Wifi_duck.ino file I get an error in the function ' void loop '
I don't know if this is the right channel to adress this error, if so please remove this issue.

Thanks in advance!
`
```
C:\Users\pyto\Desktop\sketch_may02a\sketch_may02a.ino: In function 'void loop()':

C:\Users\pyto\Desktop\sketch_may02a\sketch_may02a.ino:103:51: warning: invalid conversion from 'const char*' to 'char' [-fpermissive]

     bufferStr = ExternSerial.readStringUntil("END");

                                                   ^

In file included from C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/HardwareSerial.h:29:0,

                 from C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:232,

                 from sketch\sketch_may02a.ino.cpp:1:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Stream.h:109:10: note: initializing argument 1 of 'String Stream::readStringUntil(char)'

   String readStringUntil(char terminator);

          ^`
```

spacehuhn, 2017-05-03:
Hm do you have your board selection correct?
I don't get this error, but you should be able to fix it easily.
Change:
```
bufferStr = ExternSerial.readStringUntil("END");
```
to:
```
bufferStr = ExternSerial.readStringUntil((const char*)"END");
```

pyto00, 2017-05-04:
I am using a Pro Micro with ATMEL MEGA32U4  CPU.
in Arduino I  have " Arduino/Genuino Micro " selected. Which should be the right selection.

After altering the code I get the same error:

```
C:\Users\pyto\Desktop\sketch_may02a\sketch_may02a.ino: In function 'void loop()':

C:\Users\pyto\Desktop\sketch_may02a\sketch_may02a.ino:103:64: warning: invalid conversion from 'const char*' to 'char' [-fpermissive]

     bufferStr = ExternSerial.readStringUntil((const char*)"END");

                                                                ^

In file included from C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/HardwareSerial.h:29:0,

                 from C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:232,

                 from sketch\sketch_may02a.ino.cpp:1:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Stream.h:109:10: note: initializing argument 1 of 'String Stream::readStringUntil(char)'

   String readStringUntil(char terminator);
```

I changed it like you said:
```
void loop() {
  if(ExternSerial.available()) {
    bufferStr = ExternSerial.readStringUntil((const char*)"END");
    Serial.println(bufferStr);
  }
```

spacehuhn, 2017-05-04:
You have to use Arduino Leonardo in the board selection.

Or even better install the Pro Micro board from Sparkfun.
https://learn.sparkfun.com/tutorials/pro-micro--fio-v3-hookup-guide#installing-windows
(scroll down to ``Installing the Arduino Addon``)

pyto00, 2017-05-04:
That seemed to fix the error. 
First Used Leonardo on board selection- > restarted arduino and replugged the pro micro.
Then on verifieing it didnt give the error anymore, same for uploading.

Afterwards installed the pro micro plugin from sparkfun, no error either...

Thank you for your time!

closed 2017-05-04 by pyto00