spacehuhn/wifi_ducky #18: No Backslash "\" possible in non-US Keyboard

tomisboy, 2017-05-03:
There is no way to write a backslash,  for example in german Keyboard-Layout

I tested several thinks 

CTRL ALT ß 
did not work 
however 
CTRL ALT q 
writes the right character "@"
same with CTRL SHIRT e
which writes "€"

unfortunately alt Codes also did not work 
i tried it with 
ALT 0 9
did not work either

ahhashim, 2017-05-03:
Try Alt Gr + ß

tomisboy, 2017-05-03:
Ducky Script does not know "ALTGR"
https://github.com/hak5darren/USB-Rubber-Ducky/wiki/Duckyscript

ahhashim, 2017-05-03:
I don't know if this would help, but maybe instead of KEY_LEFT_ALT try KEY_RIGHT_ALT. Im not sure if it would be the same though.

ahhashim, 2017-05-03:
I haven't tried it on an actual Arduino board, but it compiles without errors in the Arduino IDE.

tomisboy, 2017-05-03:
I put the Code in the website (192.168.4.1)
But no KEY_RIGHT_ALT does still not work :(

spacehuhn, 2017-05-03:
OK yeah keyboard layout issues aren't really code issues, but I will look what I can do.
It's correct that at the moment there is no ALT KEY RIGHT or something implemented.
There is still a way to print that backslash, I did it before using the German keyboard layout. I'll do some tests!

Thanks for letting me know.

tomisboy, 2017-05-03:
Thank you very much and keep up your great work:)
I really appreciate all the stuff you upload and create

spacehuhn, 2017-05-04:
Ok so I think it was with the english keyboard library that I could print a backslash with a trick.
But yes I can't see how to do it with the German layout installed on the Arduino.

I will add the ALT-RIGHT key with the next update! (and also the ALT codes for windows)

tomisboy, 2017-05-04:
Awesome, thank you very much

keviiin38, 2017-05-16:
I had the same problem with the backslash, I solved it in my own issue on the Arduino forum at this address : http://forum.arduino.cc/index.php?topic=474156.0 
Hope you'll get it working !

closed 2017-05-16 by tomisboy