spacehuhn/wifi_ducky #19: Cactus Rev V.2 (Quick Question)

beezTEM, 2017-05-05:
Is there a guide to making a WiFi_Ducky with a Cactus Rev V.2?

spacehuhn, 2017-05-05:
Well the steps are the same.

**1.** Flash this onto the Atmega part (will be recognized as Lillypad Arduino USB):
https://gist.github.com/spacehuhn/b2b7d897550bc07b26da8464fa7f4b36

With this code you're able to flash the ESP8266 (it will forward the connection).

**2.** Use the 1MB bin file and upload it using the nodemcu flasher (Arduino doesn't work for that sadly):
https://github.com/spacehuhn/wifi_ducky/releases

**3.** Then you have to upload the Arduino part (same as step 1) with this sketch:
https://github.com/spacehuhn/wifi_ducky/tree/master/arduino_wifi_duck

**4.** Profit!
Your Wi-Fi Ducky is ready to use now 👍

beezTEM, 2017-05-06:
Thank you, I wasn't sure if there was anything extra you had to do with the cactus board (soldering). I know what I'm ordering soon 😄

closed 2017-05-06 by beezTEM

minkione, 2017-05-06:
checkout whid.ninja hardware is the evolution of cactus micro, madr by the same manufacturer ;)

minkione, 2017-05-07:
checkout whid.ninja hardware is the evolution of cactus micro, made by the same manufacturer ;)