spacehuhn/wifi_ducky #22: Serial and Serial1

supersjimmie, 2017-05-12:
You mention that you (cross-)connect rx/tx between the arduino and the esp8266, but in the source for arduino not Serial but Serial1 is used.
When using a 32u2 like the cjmcu beetle, is serial1 connected to the pins with rx/tx label or are those some other pins?

spacehuhn, 2017-05-12:
Serial1 means the rx/tx pins.
There are 2 'Serials' because you have to differ between the USB-Serial connection and the onboard pin-serial connection.
So yes Serial1 is for the pins, regardless of the board (as long as it's an atmega32u4 of course).

For the ESP8266 it looks a bit different. It doesn't have a native USB connection like the atmega. The Serial connection you get on your computer to, for example a NodeMCU, is made via a USB to serial chip on the board it self. This chip uses these two rx and tx pins. So for the ESP8266, there is only one serial connection for both, but the atmega has 2 seperate.

supersjimmie, 2017-05-12:
Thanks for explaining, and for adding the "question" label (wasn't able to find that myselve).
While waiting for my China 32u4 deliveries, I am testing with the 16u2 chip on my Arduino Mega2560. Do you know if the rx/tx pins on such board (which are from the also correspond to the tx/tx of the atmega2560) the same as the rx/tx pins from the 16u2?

spacehuhn, 2017-05-12:
I think so, but I'm not sure. I would say just try it out ;)

supersjimmie, 2017-05-12:
Figured it out and tested. On the Arduino Mega2560:
pin 0 (marked RX)  is TX for the 16U2 Serial1
pin 1 (marked TX)  is RX for the 16U2 Serial1

closed 2017-05-12 by supersjimmie