spacehuhn/wifi_ducky #23: Serial comm works a bit, but not good

supersjimmie, 2017-05-13:
When I simply connect the esp8266 Serial output to a serial-to-usb converter and listen to it, the whole script comes through just fine. When I connect the esp's serial output (TX) to the 16U2's serial1 input it gets corrupted.

To debug, I am just printing the 16U2's input to the usb-serial output. The code sends Serial1 in to Serial out.  For a script like this:
```DELAY 3000
GUI r
DELAY 500
STRING notepad
DELAY 500
ENTER
DELAY 750
STRING Hello World!!!
ENTER
END
```

It get this:
```DELAY 3000
GUI rL
n
0R RIo!RD0r ReL
E```

As you can see it begins normal, so baudrate shouldn't be thje problem.
I have tried several baudrates (changed on both sides) for the comm between the esp and the 16u2. 115200, 57600, 57200 (as original in your code), 38400.  It's just 2 short wires, GND-GND and TX-RX.
Any idea why it doesn't get passed correctly?

spacehuhn, 2017-05-13:
I had these errors before, try baudrate 9600.
But you have to change that for both sides of course, so make sure both the atmega and the esp script have the same baudrate setup.

supersjimmie, 2017-05-13:
A small correction on what I said, on the 16U2 I don't debug from Serial1 to Serial, but from Serial1 to Keyboard. So just:
```  
if (Serial1.available()) {
        Keyboard.write(Serial1.read());
```
Even with 9600, it;s still not good:
```DELAY 3000
GUI r
DELAY 500
STRIN oeaDLY500ETR
EA 50SRN ell ol!!ETER
ND
```

supersjimmie, 2017-05-13:
I have now added a small delay to the sending part:
```
void sendBuffer() {
  for (int i = 0; i < bc; i++) {
    Serial.write((char)scriptBuffer[i]);
    delay(2);
  }
  runLine = false;
  bc = 0;
}
```
delay(1) was not enough. delay(2) works, tested now with a baudrate of 57600 on both sides.

spacehuhn, 2017-05-13:
ha ok, so it's working fine now?

supersjimmie, 2017-05-15:
With the added delay it seems to work as a wor-around.
But I don't understand why this is needed, and it "delays" a lot...

closed 2017-06-22 by supersjimmie