spacehuhn/wifi_ducky #24: ReadStringUntil used wrong (and/or another problem reading into the buffer)

supersjimmie, 2017-05-15:
I am now having trouble reading the whole script when it comes into the 16u2.
```
ExternSerial.available()
```
Becomes true, so it starts reading into the bufferStr, but afterwards there is nothing in the bufferStr.
Not sure how to debug and then fix this...

But i noticed this: According to this https://www.arduino.cc/en/Serial/ReadStringUntil the terminator needs to be a character. In your code you specify a char*  (string like, multiple characters).

```
Warning: invalid conversion from 'const char*' to 'char' [-fpermissive]

     bufferStr = ExternSerial.readStringUntil("END");
```

marianopw, 2017-05-17:
I don't know what board are you using, but I fixed that Warning by adding `https://raw.githubusercontent.com/sparkfun/Arduino_Boards/master/IDE_Board_Manager/package_sparkfun_index.json` in Arduino > Preferences > Additional Boards Manager URLs and then selecting the proper board 'Sparkfun Pro Micro 5v' in my case

supersjimmie, 2017-05-24:
Thanks @marianopw but I am just using the 16u2 chip on an Arduino Mega2560. I don't think adding a Sparkfun Pro Micro would make a change for me. If I knew what that Sparkfun Pro Micro github code changes than perhaps I can find how to fix it in general (for all boards).

vladypunkyface, 2017-05-27:
@supersjimmie Mariano its right. I had same problem today as you. But if you follow his steps. 

"https://raw.githubusercontent.com/sparkfun/Arduino_Boards/master/IDE_Board_Manager/package_sparkfun_index.json in Arduino > Preferences > Additional Boards Manager URLs and then selecting the proper board 'Sparkfun Pro Micro 5v' in my case"

You should only change the board for a Arduino/Genuino Micro. 

I solved the problem this way.

Hope it helps

supersjimmie, 2017-06-09:
So, I installed the mentioned Boards from the board manager, selected the Sparkfun Pro Micro and the 5V 16MHz version. But still it compiles with:
```
warning: invalid conversion from 'const char*' to 'char' [-fpermissive]

     bufferStr = ExternSerial.readStringUntil("END");
```

I must add, that I have the "More" setting under my Preferences / Compiler Warnings.
Otherwise the IDE would hide most warnings and you won't see such issues.

Simply said, readStringUntil needs a char, not a char*. (like one character, not a string)

BlueArduino20, 2017-12-09:
https://github.com/spacehuhn/wifi_ducky/compare/master...BlueArduino20:patch-1