spacehuhn/wifi_ducky #30: remote fonts won't work

beamzer, 2017-05-27:
Hi Stefan,
The webserver on the esp8266 serves a skeleton.css file which contains remote fonts (raleway), but when i am connected to the esp i am not connected to the Internet anymore and the fonts won't load, resulting in delays to show the webpage. Could you remove those from the CSS?
I looked if i could do it myself, but the files are transcoded in hex.

thanks,
Ewald....

spacehuhn, 2017-05-27:
Yeah I noticed that too.
I will update the webinterface completly as soon as I get some time to do it.

If you want to change the web-files yourself, I included a little converter to get the files small and in hex, see: https://github.com/spacehuhn/wifi_ducky/tree/master/esp8266_wifi_duck/html
Will also work with the CSS file.

beamzer, 2017-05-27:
this seems to do the trick, just removed all the remote font stuff in the first part of the skeleton.cc file, debug mode on firefox confirms it.

`const uint8_t data_skeletonCSS[] PROGMEM = { 0x2e,0x63,0x6f,0x6e,0x74,0x61,0x69,0x6e,0x65,0x72,0x7b,0x70,0x6f,0x73,0x69,0x74,0x69,0x6f,0x6e,0x3a,0x72,0x65,0x6c,0x61,0x74,0x69,0x76,0x65,0x3b,0x77,0x69,0x64,0x74,0x68,0x3a,0x31,0x30,0x30,0x25,0x3b,0x6d,0x61,0x78,0x2d,0x77,0x69,0x64,0x74,0x68,0x3a,0x39,0x36,0x30,0x70,0x78,0x3b,0x6d,0x61,0x72,0x67,0x69,0x6e,0x3a,0x30,0x20,0x61,0x75,0x74,0x6f,0x3b,0x70,0x61,0x64,0x64,0x69,0x6e,0x67,0x3a,0x30,0x20,0x32,0x30,0x70,0x78,0x3b,0x62,0x6f,0x78,0x2d,0x73,0x69,0x7a,0x69,0x6e,0x67,0x3a,0x62,0x6f,0x72,0x64,0x65,0x72,0x2d,0x62,0x6f,0x78,0x7d,0x2e,0x63,0x6f,0x6c,0x75,0x6d,0x6e,0x2c,0x2e,0x63,0x6f,0x6c,0x75,0x6d,0x6e,0x73,0x7b,0x77,0x69,0x64,0x74,0x68,0x3a,0x31,0x30,0x30,0x25,0x3b,0x62,0x6f,0x78,0x2d,0x73,0x69,0x7a,0x69,0x6e,0x67,0x3a,0x62,0x6f,0x72,0x64,0x65,0x72,0x2d,0x62,0x6f,0x78,0x7d,0x40,0x6d,0x65,0x64,0x69,0x61,0x20,0x28,0x6d,0x69,0x6e,0x2d,0x77,0x69,0x64,0x74,0x68,0x3a,0x20,0x34,0x30,0x30,0x70,0x78,0x29,0x7b,0x2e,0x63,0x6f,0x6e,0x74,0x61,0x69,0x6e,0x65,0x72,0x7b,0x77,0x69,0x64,0x74,0x68,0x3a,0x38,0x35,0x25,0x3b,0x70,0x61,0x64,0x64,0x69,0x6e,0x67,0x3a,0x30,0x7d,0x7d,0x40,0x6d,0x65,0x64,0x69,0x61,0x20,0x28,0x6d,0x69,0x6e,0x2d,0x77,0x69,0x64,0x74,0x68,0x3a,0x20,0x35,0x35,0x30,0x70,0x78,0x29,0x7b,0x2e,0x63,0x6f,0x6e,0x74,0x61,0x69,0x6e,0x65,0x72,0x7b,0x77,0x69,0x64,0x74,0x68,0x3a,0x38,0x30,0x25,0x7d,0x2e,0x63,0x6f,0x6c,0x75,0x6d,0x6e,0x2c,0x2e,0x63,0x6f,0x6c,0x75,0x6d,0x6e,0x73,0x7b,0x6d,0x61,0x72,0x67,0x69,0x6e,0x2d,0x6c,0x65,0x66,0x74,0x3a,0x34,0x25,0x7d,0x2e,0x63,0x6f,0x6c,0x75,0x6d,0x6e,0x3a,0x66,0x69,0x72,0x73,0x74,0x2d,0x63,0x68,0x69,0x6c,0x64,0x2c,0x2e,0x63,0x6f,0x6c,0x75,0x6d,0x6e,0x73,0x3a,0x66,0x69,0x72,0x73,0x74,0x2d,0x63,0x68,0x69,0x6c,0x64,0x7b,0x6d,0x61,0x72,0x67,0x69,0x6e,0x2d,0x6c,0x65,0x66,0x74,0x3a,0x30,0x7d,0x2e,0x6f,0x6e,0x65,0x2e,0x63,0x6f,0x6c,0x75,0x6d,0x6e,0x2c,0x2e,0x6f,0x6e,0x65,0x2e,0x63,0x6f,0x6c,0x75,0x6d,0x6e,0x73,0x7b,0x77,0x69,0x64,0x74,0x68,0x3a,0x34,0x2e,0x36,0x36,0x36,0x36,0x36,0x36,0x36,0x36,0x36,0x36,0x37,0x25,0x7d,0x2e,0x74,0x77,0x6f,0x2e,0x63,0x6f,0x6c,0x75,0x6d,0x6e,0x73,0x7b,0x77,0x69,0x64,0x74,0x68,0x3a,0x31,0x33,0x2e,0x33,0x33,0x33,0x33,0x33,0x33,0x33,0x33,0x33,0x33,0x25,0x7d,0x2e,0x74,0x68,0x72,0x65,0x65,0x2e,0x63,0x6f,0x6c,0x75,0x6d,0x6e,0x73,0x7b,0x77,0x69,0x64,0x74,0x68,0x3a,0x32,0x32,0x25,0x7d,0x2e,0x66,0x6f,0x75,0x72,0x2e,0x63,0x6f,0x6c,0x75,0x6d,0x6e,0x73,0x7b,0x77,0x69,0x64,0x74,0x68,0x3a,0x33,0x30,0x2e,0x36,0x36,0x36,0x36,0x36,0x36,0x36,0x36,0x36,0x37,0x25,0x7d,0x2e,0x66,0x69,0x76,0x65,0x2e,0x63,0x6f,0x6c,0x75,0x6d,0x6e,0x73,0x7b,0x77,0x69,0x64,0x74,0x68,0x3a,0x33,0x39,0x2e,0x33,0x33,0x33,0x33,0x33,0x33,0x33,0x33,0x33,0x33,0x25,0x7d,0x2e,0x73,0x69,0x78,0x2e,0x63,0x6f,0x6c,0x75,0x6d,0x6e,0x73,0x7b,0x77,0x69,0x64,0x74,0x68,0x3a,0x34,0x38,0x25,0x7d,0x2e,0x73,0x65,0x76,0x65,0x6e,0x2e,0x63,0x6f,0x6c,0x75,0x6d,0x6e,0x73,0x7b,0x77,0x69,0x64,0x74,0x68,0x3a,0x35,0x36,0x2e,0x36,0x36,0x36,0x36,0x36,0x36,0x36,0x36,0x36,0x37,0x25,0x7d,0x2e,0x65,0x69,0x67,0x68,0x74,0x2e,0x63,0x6f,0x6c,0x75,0x6d,0x6e,0x73,0x7b,0x77,0x69,0x64,0x74,0x68,0x3a,0x36,0x35,0x2e,0x33,0x33,0x33,0x33,0x33,0x33,0x33,0x33,0x33,0x33,0x25,0x7d,0x2e,0x6e,0x69,0x6e,0x65,0x2e,0x63,0x6f,0x6c,0x75,0x6d,0x6e,0x73,0x7b,0x77,0x69,0x64,0x74,0x68,0x3a,0x37,0x34,0x2e,0x30,0x25,0x7d,0x2e,0x74,0x65,0x6e,0x2e,0x63,0x6f,0x6c,0x75,0x6d,0x6e,0x73,0x7b,0x77,0x69,0x64,0x74,0x68,0x3a,0x38,0x32,0x2e,0x36,0x36,0x36,0x36,0x36,0x36,0x36,0x36,0x36,0x37,0x25,0x7d,0x2e,0x65,0x6c,0x65,0x76,0x65,0x6e,0x2e,0x63,0x6f,0x6c,0x75,0x6d,0x6e,0x73,0x7b,0x77,0x69,0x64,0x74,0x68,0x3a,0x39,0x31,0x2e,0x33,0x33,0x33,0x33,0x33,0x33,0x33,0x33,0x33,0x33,0x25,0x7d,0x2e,0x74,0x77,0x65,0x6c,0x76,0x65,0x2e,0x63,0x6f,0x6c,0x75,0x6d,0x6e,0x73,0x7b,0x77,0x69,0x64,0x74,0x68,0x3a,0x31,0x30,0x30,0x25,0x3b,0x6d,0x61,0x72,0x67,0x69,0x6e,0x2d,0x6c,0x65,0x66,0x74,0x3a,0x30,0x7d,0x2e,0x6f,0x6e,0x65,0x2d,0x74,0x68,0x69,0x72,0x64,0x2e,0x63,0x6f,0x6c,0x75,0x6d,0x6e,0x7b,0x77,0x69,0x64,0x74,0x68,0x3a,0x33,0x30,0x2e,0x36,0x36,0x36,0x36,0x36,0x36,0x36,0x36,0x36,0x37,0x25,0x7d,0x2e,0x74,0x77,0x6f,0x2d,0x74,0x68,0x69,0x72,0x64,0x73,0x2e,0x63,0x6f,0x6c,0x75,0x6d,0x6e,0x7b,0x77,0x69,0x64,0x74,0x68,0x3a,0x36,0x35,0x2e,0x33,0x33,0x33,0x33,0x33,0x33,0x33,0x33,0x33,0x33,0x25,0x7d,0x2e,0x6f,0x6e,0x65,0x2d,0x68,0x61,0x6c,0x66,0x2e,0x63,0x6f,0x6c,0x75,0x6d,0x6e,0x7b,0x77,0x69,0x64,0x74,0x68,0x3a,0x34,0x38,0x25,0x7d,0x2e,0x6f,0x66,0x66,0x73,0x65,0x74,0x2d,0x62,0x79,0x2d,0x6f,0x6e,0x65,0x2e,0x63,0x6f,0x6c,0x75,0x6d,0x6e,0x2c,0x2e,0x6f,0x66,0x66,0x73,0x65,0x74,0x2d,0x62,0x79,0x2d,0x6f,0x6e,0x65,0x2e,0x63,0x6f,0x6c,0x75,0x6d,0x6e,0x73,0x7b,0x6d,0x61,0x72,0x67,0x69,0x6e,0x2d,0x6c,0x65,0x66,0x74,0x3a,0x38,0x2e,0x36,0x36,0x36,0x36,0x36,0x36,0x36,0x36,0x36,0x36,0x37,0x25,0x7d,0x2e,0x6f,0x66,0x66,0x73,0x65,0x74,0x2d,0x62,0x79,0x2d,0x74,0x77,0x6f,0x2e,0x63,0x6f,0x6c,0x75,0x6d,0x6e,0x2c,0x2e,0x6f,0x66,0x66,0x73,0x65,0x74,0x2d,0x62,0x79,0x2d,0x74,0x77,0x6f,0x2e,0x63,0x6f,0x6c,0x75,0x6d,0x6e,0x73,0x7b,0x6d,0x61,0x72,0x67,0x69,0x6e,0x2d,0x6c,0x65,0x66,0x74,0x3a,0x31,0x37,0x2e,0x33,0x33,0x33,0x33,0x33,0x33,0x33,0x33,0x33,0x33,0x25,0x7d,0x2e,0x6f,0x66,0x66,0x73,0x65,0x74,0x2d,0x62,0x79,0x2d,0x74,0x68,0x72,0x65,0x65,0x2e,0x63,0x6f,0x6c,0x75,0x6d,0x6e,0x2c,0x2e,0x6f,0x66,0x66,0x73,0x65,0x74,0x2d,0x62,0x79,0x2d,0x74,0x68,0x72,0x65,0x65,0x2e,0x63,0x6f,0x6c,0x75,0x6d,0x6e,0x73,0x7b,0x6d,0x61,0x72,0x67,0x69,0x6 [... cut ...] b,0x62,0x6f,0x72,0x64,0x65,0x72,0x2d,0x74,0x6f,0x70,0x3a,0x31,0x70,0x78,0x20,0x73,0x6f,0x6c,0x69,0x64,0x20,0x23,0x45,0x31,0x45,0x31,0x45,0x31,0x7d,0x2e,0x63,0x6f,0x6e,0x74,0x61,0x69,0x6e,0x65,0x72,0x3a,0x61,0x66,0x74,0x65,0x72,0x2c,0x2e,0x72,0x6f,0x77,0x3a,0x61,0x66,0x74,0x65,0x72,0x2c,0x2e,0x75,0x2d,0x63,0x66,0x7b,0x63,0x6f,0x6e,0x74,0x65,0x6e,0x74,0x3a,0x22,0x22,0x3b,0x64,0x69,0x73,0x70,0x6c,0x61,0x79,0x3a,0x74,0x61,0x62,0x6c,0x65,0x3b,0x63,0x6c,0x65,0x61,0x72,0x3a,0x62,0x6f,0x74,0x68,0x7d };`

supersjimmie, 2017-08-18:
@beamzer  or @spacehuhn can you confirm that this modified skeleton fixes the issue?
I have my hardware built-in and I am currently using an esp01 which cannot be updated via the web-interface. So making a change needs opening my device and re-wiring the serial etcetera. So I would like to know this "for sure" before going to this change...

supersjimmie, 2017-08-18:
Found some time to do it, and it seems to be working!

Mr-iX, 2017-08-28:
Could you please test if the Pull Request #64 fixes the issue for you?

careyer, 2017-08-31:
@Mr-iX: Yeahii seems to work! ;-) The different pages now also load nicely on a mobile phone without internet connection! Unfortunatly I had to desolder my Wifi Ducky in order to reflash the ESP8266... flashing via the Webinterface did somehow not work (can someone please decribe how that should work? Do I need to put GPIO_0 to LOW? At what time (boot time, just short it out to GND briefly?)?