spacehuhn/wifi_ducky #31: problem with P

beamzer, 2017-05-28:
I made a build with a wemos d1 mini, the RX/TX pins align mirrored with those of the CJMCU  atmega32u4, so i could use two header pins and solder the boards together and then only use two wires for 5V and Ground (the Wemos has a power regulator).

It kind of works, but i found two problems so far and can't imagine that those are because of my hardware setup?

One is that the capital P doesn't get output. And when i send:
STRING ABCDEFGHIJKLMNOPQRSTUVWXYZ
i get:
ABCDEFGHIJKLMNO
lower case there is no problem, and i did not find any other characters yet which "misfire".

The other thing is that if i send 
SPACE
it just seems to send capital S
and then it seems to hit the "P" problem

any ideas?
(should probably try with a lower baudrate, but now i have to desolder every time i want to program the CJMCU atmega32u4 :(  and if it was a baudrate problem i would expect more problems)

[image]

tobozo, 2017-05-31:
This is why I've soldered a switch on my build

[image]

beamzer, 2017-06-01:
nice setup for testing.
So do SPACE and P just work in your case?

tobozo, 2017-06-02:
Yup both work fine (BAUD_RATE=115200).

Did you observe this SPACE/P truncation on all browsers?

beamzer, 2017-06-10:
Found it, i used the BAUD_RATE in spacehuhn's source which is 57200, this is 400bps off from the standard baudrate of 57600. When i switched to 57600 on both devices, my problem with P was gone ;) Will now try 115200 and if it works consistently stick to that.
It might have to do with production, firmware or tolerances. I had a different device from aliexpress which worked fine (as far as i could tell) on 57200, but 57600 should work for anything.

beamzer, 2017-06-10:
hmm, cheered to soon, it's back even with 57600 baud  😢 still investigating

careyer, 2017-09-24:
I solved the problem by reducing the baud rate on both the Teensy as well as the ESP8266 to 19200.  See #65