spacehuhn/wifi_ducky #34: Scripts not being sent through

Dev01-D, 2017-06-07:
Managed to get wifi ducky to broadcast an SSID and connect to it on 192.168.4.1 but non of the 'live execute' scripts i am trying are working. even tried the basic:
GUI r
STRING regedit
ENTER

beamzer, 2017-06-10:
I found that some devices don't like the 57200 baudrate in the source (see my issue: *Problem
with P* https://github.com/spacehuhn/wifi_ducky/issues/31 ) this might also be the cause for your scripts not comming through? Switch to 57600 and see it that helps

Otherwise, you can enable debug on the esp software (for some reason the variable is both in esp8266_wifi_duck and settings.h) and connect that to second computer and watch what's happening over USB-serial. But **separate the power** if your doing that, because both devices will be fed 5V over USB. So you would only use GNB, RX and TX between them

br0k3nilluzion, 2017-06-27:
I am getting the same issue. I can connect to AP, but scripts will not Live Execute.  
Is there a known problem with Windows 7 and this setup? I am using a Pro Micro and D1 mini as well