spacehuhn/wifi_ducky #35: Improvement suggestion: SoftwareSerial on the esp

supersjimmie, 2017-06-09:
Everytime I want to change something on the esp8266, I need to disconnect it's RX/TX from the arduino.
Perhaps we can just change the esp code to use softwareserial on different pins, so it's real rx/tx can be kept free for (re-)programming:

```
#define SERIAL_TX       12
#define SERIAL_RX       14
#include <SoftwareSerial.h>  // https://github.com/plerup/espsoftwareserial
SoftwareSerial mySerial(SERIAL_RX, SERIAL_TX, false, 255); // (rxpin, txpin, or -1, inverted, buffersize)
```
And then change each ```Serial``` to ```mySerial```.

spacehuhn, 2017-06-09:
In my experience is softwareserial not really reliable.
And we need something reliable and fast here. Depending on the hardware choices you already get into problems using the normal serial pins.

But it's a good idea for an option in the settings!

supersjimmie, 2017-06-09:
I think the most important side for the "fast and reliable" is the receiving side.
Here I am talking about the esp8266, which is the sender. The esp8266 has no problems at all dealing with SoftwareSerial, I'm using it in other projects without any issue.

I do agree that the reveiving part, the 32u4, is more sensitive. There, we need more reliable "hardware" serial. But because that one already has it's CDC Serial and Hardware Serial1, that's not an issue.

supersjimmie, 2017-06-10:
By the way, don't compare the quality of the SoftwareSerial for the esp8266 with the SofwareSerial(-INT10) for the DigiSpark modules.  I know the one for DigiSpark isn't working very well, and I also know that the esp's version is very good.

Actually, I have it working here with your code and it's totally fine.  :)

beamzer, 2017-06-10:
If you want to update the esp8266 without disconnecting the arduino(-clone),
you can just use the *over the air update* as mentioned in Spacehun's readme:
https://github.com/spacehuhn/wifi_ducky#update-esp8266-over-the-webinterface

Only when you need to reprogram the arduino, you need to decouple the devices,
but the arduino code is really basic, so chances are that you will never have to update that.

supersjimmie, 2017-06-10:
I know we can update OTA, if the esp is operational. But when I am developing new code that sometimes could fail, it's nice to have an easy way to do it over serial. Also debugging is much easier when the serial for the esp-arduino is separated from the esp's debugging serial.

It's just an idea. 
Adding one #include line, changing one Serial.write line, and wiring rx/tx to 2 other pins. It's that easy.
If you start new, it doesn't even make any difference at all...

spacehuhn, 2017-06-25:
I tested the Software Serial and atleast for my setup it wasn't usable.
But I found a better way.

So when debug is on the ESP sends messages which the ATmega wants to interpret as Ducky Script, so I added a prefix to all debug messages `:: this is a debug message`. When the ATmega gets these messages it will print them but not execute them further.

And before you guys ask, I promise I will update everything next month. I don't have much time at the moment.