spacehuhn/wifi_ducky #39: Not working

D0V15, 2017-06-20:
I did everything according to Seytonic's instructions (https://www.youtube.com/watch?v=Utq4C9S3-uI) accept I used the NodeMcu instead of WeMos d1 mini. So i plug in my wifi duck and it automatically starts writing J88888888888888888888J888888(without stopping unless I plug it out)
And I don't understand why it is happening could anyone help me? Thanks!

tobozo, 2017-06-21:
Check your baud rate, on both sides it should be 57600, not 57200. This does the trick for me on every build. Also see #34 #23 and #31 where other symptos seem to be triggered by the same situation.

@spacehuhn is this 57200 baud rate a typo or is it intentional ?

spacehuhn, 2017-06-21:
@tobozo 
I'm actually not sure. I will go back to the 115200 anyway. The 52*00 baudrate was only to support the cactus and whid board. But I don't care about them anymore, if Aprbrother is unable to put a crystal on there that is good enough for that standard baudrate, it's their problem!
I don't want to use 9600, that would be super slow. 115200 is standard for the ESP and works very well.

I will start working on an update soon. I just have a lot of other things to do at the moment.

D0V15, 2017-06-21:
@tobozo I tried lowering the baud rate to 57600 and it did nothing everything stayed like before and i forgot to stay that 2 LED's turn on when pluged in(I think there is only one supposed to turn on?)
[image]
But still thanks dor the help and @spacehuhn waiting for the new update!

tobozo, 2017-06-21:
@D0V15 can you post a pic of the full assembly w/readable pinout (unplugged) ?

D0V15, 2017-06-22:
[image]
[image]
[image]

tobozo, 2017-06-22:
Everything looks right, could be a crappy voltage regulator on the nodeMCU

Other ways to power the NodeMCU:

Arduino       NodeMCU
------------------------
VCC            vin       (your current setup)
Raw             vin       (what you could try)

And only if your Arduino is the 3.3v version:

Arduino       NodeMCU
------------------------
VCC            3v        
Raw             3v        (only if your Arduino is the 3.3v version !!)

supersjimmie, 2017-06-22:
Based on those images, you power the LoLin esp8266 from the Arduino Pro Micro VCC pin.
Is the Arduino a 3.3V or 5V type?  

If the Arduino is a 3.3V type, the VCC is feeding that 3.3V into the Vin, which needs 5V on the LoLin.  The Vin goes to the onboard voltage regulator and needs at least 5V as Vin.

If the Arduino is a 5V version, you have 2 other problems:
1. The Arduino VCC will not be able to feed enough current for the LoLin,
2. The Arduino TX puts a 5V signal on the LoLin RX, which is a 3.3V input pin. This can kill it.

My advice: 
1. connect the LoLin Vin to the Arduino RAW, that way the LoLin is powered directly from the USB bus (the LoLin Vin goes through a 3.3V regulator, which needs a 5V input).
2. If the Arduino is a 5V version, use a 5V-to-3.3V levelshifter to connect it's TX to the LoLin RX.

spacehuhn, 2017-06-22:
> The Arduino TX puts a 5V signal on the LoLin RX, which is a 3.3V input pin. This can kill it.

The ESP8266 has a 5V protection on the IO pins. I know there is an on-going discussion because Espressif won't say that officially but it's safe up to 12mA, if I'm right. So one serial pin running on 5V is absolutely no problem.

D0V15, 2017-06-22:
@tobozo it kinda fixed it is now working almost properly now it has the same problem as #34

supersjimmie, 2017-06-22:
So change both sides to 57600 or 115200?

D0V15, 2017-06-22:
@tobozo  @supersjimmie  Thanks! It's working now(a little slow because the baud on nodemcu is 57600) But still huge thanks!

closed 2017-06-22 by D0V15

spacehuhn, 2017-06-25:
Can you guys maybe test if 115200 works for you?
I never got problems using that, but I have a 3.3V Pro Micro setup here that won't work. Maybe it is depending on the ATmega's clocking (3V: 8MHz, 5V: 16MHz).

tobozo, 2017-06-26:
I'll give it a try this week, here's what I've observed so far on WifiDucky and on an Arduino/U-Blox-GPS:

- 3.3V;8MHh get more serial errors over 57600 bauds. Possible explanation: the closest clock divider ratio for 115200 baud is slightly out of range when generated from a 8MHz clock. And YMMV depending on the quality of the hardware (i.e. shitty clones).
- Setting serial port configuration to 8n2 (2 stop bits) `Serial.begin(115200, SERIAL_8N2);` mitigates this effect on some models. 

If I'm out of spare Arduinos, I can use a level shifter as a last resort, otherwise I just 🗑 the faulty unit.