spacehuhn/wifi_ducky #40: Just whowing

supersjimmie, 2017-06-22:
Here's my result:
[image]

supersjimmie, 2017-06-27:
The 2 buttons and the 3-pin header are just extra's and can be removed.

BlueArduino20, 2017-08-23:
Hi, I've got some questions about your creation. Have you got Discord account or... email?

0xCoto, 2017-08-23:
[image]

I win. ;)

BlueArduino20, 2017-08-23:
Oh, man. You evil guy.

Mr-iX, 2017-08-24:
@0xCoto Could you share, how you made this?

0xCoto, 2017-08-26:
@Mr-iX Connected all the Pins, uploaded the USB-to-Serial Arduino Code to the SS Micro, flashed the ESP-12E with the esp-flasher.exe and uploaded the Arduino-Ducky code to the SS Micro.

tobozo, 2017-08-26:
So this build solves the problem of not being able to flash after soldering, am I right ?

Mr-iX, 2017-08-26:
[image]

@0xCoto 
Is that the right circuit diagram for this build?
Where have you hidden all the required cables?

BlueArduino20, 2017-08-26:
I think your diagram is completely wrong.
He have hidden the cables under the ESP-12E, there is enough space.

Mr-iX, 2017-08-26:
@BlueArduino20 I just draw the diagram side-by-side only for simplicity.
You have to move the right side (ESP-12E) over the left side (Arduino SS Micro), to have the real construction.

0xCoto, 2017-08-26:
@Mr-iX Then why are you asking where I hid the cables?

And no, that is not the correct diagram.

Mr-iX, 2017-08-26:
@0xCoto I haven't known if there is enough space. But now I know ;)
What is wrong with the diagram? I thought this would be right for flashing the ESP module.

0xCoto, 2017-08-26:
These are the correct connections:
TX to RX
RX to TX
3.3v to VCC
GND to GND
GND to GPIO15
D14 to GPIO0
D15 to EN

BlueArduino20, 2017-08-26:
@Mr-iX I wasn't saying that the cables are the problem of the diagram.

Mr-iX, 2017-08-26:
Now this should be the correct one. Maybe it helps someone in the future.
[image]

@BlueArduino20 Sorry I understood you wrong

BlueArduino20, 2017-08-26:
Yeah, that should be the right diagram.
Also you need to modify the first two lines of the serial code.
From:
<pre><code>int program_pin = 12;
int enable_pin = 13;</pre></code>
To:
<pre><code>int program_pin = 14;
int enable_pin = 15;</pre></code>

supersjimmie, 2017-08-30:
@BlueArduino20  what do you mean you have a question and want my mail?

@0xCoto yeah you win with that picture.  but I have now a complete working combination of both @spacehuhn **Keylogger** _and_ @spacehuhn **wifi_ducky** together in one code and one piece of hardware built into an HP keyboard.
I could place a picture here, but that would just be a pic of an HP keyboard.  :)
So here's the keyboard opened:
[image]

This nifty thing is capable of logging keystrokes, making them visible over it's own hidden wifi, and then attacking with a Ducky-script over that wifi again.  I made a small demo here with my friends with a clean W10 pc. The pc was not connected to any network. One person was asked to change the local password and lock the pc. Then I took that password out of the keylogger-data and placed it into a Ducky script that unlocked the pc. After unlocking another Ducky script started notepad and typed the text:
```
You have been Ducked.
Your password is...  (showing the password here)
```

Even after I told them that is was "some kind of hardware" it took them a lot of time and several hints to discover it.

0xCoto, 2017-08-30:
@supersjimmie That's awesome! I would love to learn more, since Spacehuhn's Keylogger is pretty large and not as incognito as this ^. Been trying to make such hardware for a long time. What's your Discord/Skype/Twitter?

careyer, 2017-08-31:
@BlueArduino20 : Where can this Serial-Code be found?
int program_pin = 14;
int enable_pin = 15;

Cannot find these variables in any of those files?
Thanks!

0xCoto, 2017-09-01:
@careyer https://gist.github.com/spacehuhn/b2b7d897550bc07b26da8464fa7f4b36

careyer, 2017-09-04:
@0xCoto thank you very much.... i did not realize that I had to flash a completely different sketch to the arduino ;-). I thought that there must be a way to do it via the Update function in the GUI (Info Tab --> Update).

supersjimmie, 2017-09-05:
@0xCoto just use my Github name at gmail dot com
(I'm not reading that account daily though)

br0k3nilluzion, 2017-09-06:
@supersjimmie are you going to share that bby on your github?

0xCoto, 2017-09-06:
@supersjimmie FIY I sent you a mail.

supersjimmie, 2017-09-07:
@br0k3nilluzion would love to do so, but I just started at my new job so I don't have much time.
But I did start thinking about it...
@0xCoto replied.  ;)

br0k3nilluzion, 2017-09-17:
@supersjimmie yeah I feel you on that.. all work and no play.
Would love to see it when you get a chance to share.

GarethWright, 2017-09-18:
I second that @br0k3nilluzion, @supersjimmie is exactly....down to the keyboard what I'm going for

Mr-iX, 2017-09-22:
Now I have built the WiFi Ducky from @0xCoto.
I have connected all the wires (picture above) and uploaded this to the Arduino:
`int program_pin = 14;
int enable_pin = 15;

void setup()
{
  Serial1.begin(115200);
  Serial.begin(115200);
  pinMode(enable_pin, OUTPUT);
  pinMode(program_pin, OUTPUT);
  digitalWrite(program_pin, LOW);
  digitalWrite(enable_pin,HIGH);
}

void loop()
{
  while(Serial1.available()){
    Serial.write((uint8_t)Serial1.read());
  }

  if(Serial.available()){
    while(Serial.available()){
      Serial1.write((uint8_t)Serial.read());
    }
  }
}`

But when I try to flash the esp8266_wifi_duck_4mb.bin file to the ESP8266, I get following error message:
`sh-3.2# esptool.py --port /dev/cu.usbmodem3A21 write_flash -fm dio 0x00000 esp8266_wifi_duck_4mb.bin 
esptool.py v2.0.1
Connecting........_____....._____....._____....._____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header`

I have tried a second ESP and Arduino, but I get the same error.
Can you give me a hint what I'm doing wrong?

0xCoto, 2017-09-23:
@Mr-iX Have you tried with the esp-flasher.exe? It worked for me.

Mr-iX, 2017-09-23:
@0xCoto You mean that tool?
https://github.com/nodemcu/nodemcu-flasher

It worked with that, but after all steps, I cannot find a WiFi Network

BlueArduino20, 2017-09-24:
@Mr-iX You have to modify the "**arduino_wifi_duck.ino**" file too.
**From:**
<pre><code>  pinMode(13,OUTPUT);
  digitalWrite(13,HIGH);
</pre></code>

**To:**
<pre><code>  pinMode(15,OUTPUT);
  digitalWrite(15,HIGH);
</pre></code>

Mr-iX, 2017-09-24:
@BlueArduino20 Thank you so much, that was the solution!

careyer, 2017-09-24:
I have re-wired my hardware according to @0xCoto suggested (and @Mr-iX depicted). 
It does not work with the original code though. You have to alter the following lines in the arduino_wifi_duck.ino code:

`  pinMode(13,OUTPUT); 
  digitalWrite(13,HIGH); `

to 

`  pinMode(15,OUTPUT); 
  digitalWrite(15,HIGH); `

otherwise the Enable-PIN will not be set HIGH and the ESP8266 will not be activated.

careyer, 2017-10-13:
@supersjimmie: more and more I like the idea of your combined keylogger+ducky solution. Would you mind to share the combined code and/or document how you build the hardware? That HP keyboard looks neat ;-)

supersjimmie, 2017-11-06:
@careyer see https://github.com/supersjimmie/wifi_ducky_keylogger
The code is there, only missing the readme.md.

careyer, 2017-11-07:
@supersjimmie : Great thank you very much! Could you please also add a short description on how to wire things up? Just giving very basic direction! That would really help! Thank you very much!

supersjimmie, 2017-11-07:
@careyer Sure I can...  (mubles something about time...)

supersjimmie, 2017-11-07:
Ah well.. Found just enough time to add the basics.
Let me know how it goes with that?

careyer, 2017-11-08:
@supersjimmie : Thank you very much! That makes things much clearer! =) Well done! I will give it a shot! :)