spacehuhn/wifi_ducky #41: Some web interfaces issues

supersjimmie, 2017-06-23:
From the IDE, I select Sketch - Export compiled Binary.
The .bin file is created in the sketch folder and has the same date/time from the export moment.
It's size is about 134kb.
Then I import it in the web interface under 192.168.4.1/update, but the web interface returns "FAIL".

When I upload a script and then press the Run button, it runs fine.
When I go to the page to edit it, or create a new one from the web interface, and press the Run button there, it does not run.  (while it runs fine again when I go back to the main Run button).

supersjimmie, 2017-06-26:
ESP12-F, Chrome on Windows 10.

tobozo, 2017-06-26:
What model are you using in the tools menu of your Arduino IDE ?
If using Generic-ESP8266, check that the flash size is corresponding to your hardware.

supersjimmie, 2017-06-26:
Yes, I was thinking in that direction too.
Yesterday I was strugling to get spiffs working on another esp01 module, that too had to do with the sizes.

I'll check the actual and configured values as soon as I can work on it again.
CheckFlashConfig.ino was my friend before...

spacehuhn, 2017-06-26:
I got into probelsm too when I wanted to update a 1mb(512kb SPIFFS) ESP module. I'm not sure if this configuration supports flashing it that way, because it probably has a limited amount of reserved memory for this feature. And I never had problems with the 4mb modules.

supersjimmie, 2017-06-27:
I've read that the 1MB versions are unable to do OTA or browser updating. (can't find the source now)
But even with a 4MB version that is incorrectly set in the IDE as 1MB, it won't work.

So, while I haven't been able to test this, I think the issue is that I have a 4MB esp with 1MB selected in the IDE.

tobozo, 2017-06-27:
Yep 1MB is not enough to do OTA since your binary size can't exceed half the size of available space, it fails as expected.

Have you tried to flash it with "NodeMCU 1.0 ESP-12E Module" or "Wemos Mini D1" selected in the hardware menu ? 

Those cards have presets that are know to work well with this project. If it still fail with those settings, then your ESP-12F is probably a 1MB version and can't do OTA.

K1two2, 2017-06-27:
I use CJMCU-Beetle ATMEGA32U4 + ESP8266-12F successful operation, and now I would like to compile the Chinese web interface, I would like to ask arduino IDE how I choose the development board, as well as the development board configuration, please give me some pictures, thank you ~

supersjimmie, 2017-06-27:
@K1two2 
For the CJMCU Beetle you can use the board Arduino/Genuino Micro, or the Arduino Leonardo.
For the ESP12F you can use the board Generic ESP8266 Module, then set Flash Size to 4M (1M SPIFFS).

K1two2, 2017-06-27:
I modified the HTML and tested normal on the PC. Build firmware and upload updates via the web. But the code in Chinese shows the code. What is the reason for this? I checked the code of HTML is utf-8

supersjimmie, 2017-06-27:
@K1two2 
I would suggest that you open a new issue for this, instead of using this topic.

supersjimmie, 2017-07-07:
Sorry I didn't have time before...
I have re-uploaded the sketch to my esp12-f with the correct 4MB (1MB spiffs) and now a web update works fine.  

So not only having 4MB is enough, you also need the Arduino IDE to be set correctly to that.
The CheckFlashConfig.ino file can be very helpful: https://github.com/esp8266/Arduino/blob/master/libraries/esp8266/examples/CheckFlashConfig/CheckFlashConfig.ino

closed 2017-08-18 by supersjimmie