spacehuhn/wifi_ducky #42: not an issue, but I dont know where to ask! will this work on a nodemcu, it does have an esp8266, but I cannot find the type of processor

enderface36, 2017-06-30:


spacehuhn, 2017-07-01:
Yes it will work. The ESP8266 is the processor (or SoC to be exact). And the module is an ESP-12E or F, but which one doesn't matter.

SkyCityCZ, 2017-09-11:
Hello i have the esp8266 (the deauther video) http://www.ebay.com/itm/D1-Mini-NodeMcu-4M-bytes-Lua-WIFI-Development-Board-ESP8266-by-WeMos-/182035363514?epid=514207412&hash=item2a622736ba:g:NSYAAOSwe7BWzu4m&siteId=0&pguid=475465c615e0ac8007e2d2dbfd365a35&AdChoicePreference=true&rmvSB=true is a way to make it only with this?

Jamezrin, 2017-09-11:
> Hello i have the esp8266 (the deauther video) http://www.ebay.com/itm/D1-
> Mini-NodeMcu-4M-bytes-Lua-WIFI-Development-Board-ESP8266-by-WeMos-/
> 182035363514?epid=514207412&hash=item2a622736ba:g:
> NSYAAOSwe7BWzu4m&siteId=0&pguid=475465c615e0ac8007e2d2dbfd365a
> 35&AdChoicePreference=true&rmvSB=true is a way to make it only with this?

That is a esp8266, you can just flash that

SkyCityCZ, 2017-09-11:
It doesnt do anything
the ui works i go to 192.168.4.1 and  can see the website but if i try live execute like GUI it does nothing

supersjimmie, 2017-09-13:
@SkyCityCZ  I must say that on my setup that "Live" does not work too.
Most other functions, like upload and then Run work fine.
So please continue and try to upload a script (make a .txt file and and upload that) and click the Run button to see what that does?

hacker4313, 2018-04-02:
i have nodemcu flashed bin file when i go to 192.168.4.1  all works perfectly fine but when i upload script its not working please help 🤐😥😭

777anon666, 2018-04-11:
to all the people having issue getting WiFi Ducky to work it could be a issue with the .bin you compiled or are using i was having the same issue but i was able to fix it by editing and compiling my own WiFi Ducky ESP .bin file I have the Whid.ninja Cactus WHID 4m.... if that is the same hardware you are having issues getting WiFi Ducky to work on email me and i will email you my custom .bin to flash and try Email: 777anon666@protonmail.com