spacehuhn/wifi_ducky #43: can't compile "arduino_wifi_duck"

BUDDIKA98, 2017-07-06:
`Arduino: 1.6.9 (Windows 10), Board: "Arduino/Genuino Uno"

C:\Users\Pinsara udara\Desktop\New Folder\wifi_ducky-1.0\arduino_wifi_duck\arduino_wifi_duck.ino: In function 'void Line(String)':

arduino_wifi_duck:12: error: 'Keyboard' not found. Does your sketch include the line '#include <Keyboard.h>'?
     for(int i=firstSpace+1;i<_line.length();i++) Keyboard.write(_line[i]);

                                                  ^

arduino_wifi_duck:38: error: 'Keyboard' not found. Does your sketch include the line '#include <Keyboard.h>'?
   Keyboard.releaseAll();

   ^

C:\Users\Pinsara udara\Desktop\New Folder\wifi_ducky-1.0\arduino_wifi_duck\arduino_wifi_duck.ino: In function 'void Press(String)':

arduino_wifi_duck:44: error: 'Keyboard' not found. Does your sketch include the line '#include <Keyboard.h>'?
   if(b.length() == 1) Keyboard.press(char(b[0]));

                       ^

arduino_wifi_duck:45: error: 'Keyboard' not found. Does your sketch include the line '#include <Keyboard.h>'?
   else if (b.equals("ENTER")) Keyboard.press(KEY_RETURN);

                               ^

arduino_wifi_duck:45: error: 'KEY_RETURN' was not declared in this scope

   else if (b.equals("ENTER")) Keyboard.press(KEY_RETURN);

                                              ^

arduino_wifi_duck:46: error: 'Keyboard' not found. Does your sketch include the line '#include <Keyboard.h>'?
   else if (b.equals("CTRL")) Keyboard.press(KEY_LEFT_CTRL);

                              ^

arduino_wifi_duck:46: error: 'KEY_LEFT_CTRL' was not declared in this scope

   else if (b.equals("CTRL")) Keyboard.press(KEY_LEFT_CTRL);

                                             ^

arduino_wifi_duck:47: error: 'Keyboard' not found. Does your sketch include the line '#include <Keyboard.h>'?
   else if (b.equals("SHIFT")) Keyboard.press(KEY_LEFT_SHIFT);

                               ^

arduino_wifi_duck:47: error: 'KEY_LEFT_SHIFT' was not declared in this scope

   else if (b.equals("SHIFT")) Keyboard.press(KEY_LEFT_SHIFT);

                                              ^

arduino_wifi_duck:48: error: 'Keyboard' not found. Does your sketch include the line '#include <Keyboard.h>'?
   else if (b.equals("ALT")) Keyboard.press(KEY_LEFT_ALT);

                             ^

arduino_wifi_duck:48: error: 'KEY_LEFT_ALT' was not declared in this scope

   else if (b.equals("ALT")) Keyboard.press(KEY_LEFT_ALT);

                                            ^

arduino_wifi_duck:49: error: 'Keyboard' not found. Does your sketch include the line '#include <Keyboard.h>'?
   else if (b.equals("GUI")) Keyboard.press(KEY_LEFT_GUI);

                             ^

[... 314 lines cut ...]

   if(ExternSerial.available() > 0) {

      ^

arduino_wifi_duck:3: error: 'Serial1' was not declared in this scope

 #define ExternSerial Serial1

                      ^

C:\Users\Pinsara udara\Desktop\New Folder\wifi_ducky-1.0\arduino_wifi_duck\arduino_wifi_duck.ino:112:5: note: in expansion of macro 'ExternSerial'

     ExternSerial.write(0x99);

     ^

exit status 1
'Keyboard' not found. Does your sketch include the line '#include <Keyboard.h>'?

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`
how to fix that?

tobozo, 2017-07-06:
Use your Arduino Library Manager and install the library "Keyboard" as hinted in the first error message

BUDDIKA98, 2017-07-06:
I install the "Keyboard" library. But not fix. how to fix that?

tobozo, 2017-07-06:
This library?  Don't forget to restart your Arduino IDE after installation.

[image]

BUDDIKA98, 2017-07-06:
Not working

BUDDIKA98, 2017-07-06:
Ok. Now it work. That's not work with Arduino uno

closed 2017-07-06 by BUDDIKA98

tobozo, 2017-07-06:
Maybe select a different board ? You did not specify your hardware in your problem description and it seems like you're compiling with Uno board, which has no 32u4/USB host.