spacehuhn/wifi_ducky #61: To do list

harshil2105, 2017-08-10:
I'm interested in your projects and i see that you have a cool "To-Do" list....can you make the appearance and other stuff better in this project too just as you did in esp8266 deauther.....Do you have any update on this project or this is a abandoned project?

spacehuhn, 2017-08-10:
It's not abandoned, it just needs a lot of work and I don't have the time at the moment. 
I will continue developing it soon.

harshil2105, 2017-08-10:
It's so nice of you....WiFi 🦆 is one of the best project you have made
.....we just expect you to make some changes in this project too,I'm very
excited to see what you do next

jaime29010, 2017-08-10:
You know, anyone can contribute, not only @spacehuhn has to do stuff.
It's funny because I never contributed to anything here xD

harshil2105, 2017-08-11:
It's a cool project.... make some changes, best regards

SkyCityCZ, 2017-11-03:
To do: Remove the 1024 limit

supersjimmie, 2017-11-06:
@SkyCityCZ what 1024 limit?

SkyCityCZ, 2017-11-06:
For wifi ducky scripts...

supersjimmie, 2017-11-07:
I don't have such large scripts, and if you simply break one >1024 sized script into more smaller scripts you can send as much as you want.
There is a var called bufferSize in the esp part of the code, which is now set to 600. That would suggest that the limit is not even 1024 but only 600.  You could experiment with that.

But beside that bufferSize, perhaps it's just a limitation on the serial hardware or another var that is rolling over.  Actually, you should be able to find out at which point you start loosing data by simply adding some debugging lines to your own code.