spacehuhn/wifi_ducky #63: Take a look to this 3,3V Beetle

careyer, 2017-08-27:
Sorry, no issue but wanted to let you know:
http://www.ebay.de/itm/Beetle-Leonardo-ATMEGA32U4-Mini-Development-Board-Arduino-kompatibel-296/321914373613?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

It is 3,3V ready out of the box ;-)
Cheers!
Keep up the good work

careyer, 2017-08-29:
Got one of those boards shipped today.
It shows up as a "Teensy USB Serial (COM24)" in the device manager.

I was able to flash the arduino_wifi_duck.ino sketch to it (COM24, Arduino Leonardo), after fixing a minor problem in codeline 103:
`
102  if(ExternSerial.available()) {
103   bufferStr = ExternSerial.readStringUntil("END");  // it should be 'END' instead of "END", right?
104 Serial.println(bufferStr);
`

Upload succeeded and I confirmed the sketch runs. However the device still is detected as "Teensy USB Serial". Shouldn't it be deteced as a keyboard or HID device?
I have not yet wired up the ESP8266 since it was not shipped yet. I am just curious if there is a problem with that Teensy stuff and that it does not show up as a HID

Thanks in advance for helping me here!

supersjimmie, 2017-08-30:
For your compile issue, see: https://github.com/spacehuhn/wifi_ducky/issues/24

Yes even after uploading the code it will still (also) be seen as the original device. Otherwise you wouldn't be able to upload any new code anymore. I guess there may be some hacks to remove the USB Serial afterwards, but as I said that would make your device unable to be reprogrammed.

Anywa, nice discovery of a 3.3V version! (but I think it's very expensive, more than twice the price compared to the 5V)

careyer, 2017-08-30:
Thank you! Yes, the 3,3V version is a bit more pricey - however it was delivered right next day! I buy the offset in price for this premium. It also makes the whole unit very slim an compact. 
Did the solution in #24 solve the issue for you? You mentioned in the end that you would still get the compiler warning? Cheers!

supersjimmie, 2017-08-30:
Now I take a closer look at your 3.3V version, it looks exactly the same as my 5V. Tey have both a 3.3V and a 5V connection. Also on your link the information says it's 5V operation.  This is the one I have: https://nl.aliexpress.com/item/Free-shipping-SS-micro-ATMEGA32U4-module-compatible-for-arduino-pro-micro/32660718568.html

I know there is a discussion whether the esp8266 can handle 5V on it's inputs. All I can say is that I never had problems connecting a 3.3V esp and a 5V arduino.

About the Issue #24 it is still giving me a warning like in the open issue, but all seems to be working so I simply ignore it.  It's all just hobby so if it works, it's fine.

careyer, 2017-08-30:
Haha! =D ... Yes the Arduino (the CPU) works at 5V. However the big voltage regulator on the backside should be a 3,3V regulator. Have you tried to measure the voltage between that Pin marked with 3,3V and GND? If it is indeed 3,3V (which I assume it is) you should be safe coupling your Arduino and ESP together without an additional regulator ;-)

careyer, 2017-08-31:
Here we go! Small, neat and clean ;-)
[image]
Perfect - if only there wasn't that #65 problem.

gloglas, 2018-02-13:
It is named 'Micro SS'