spacehuhn/wifi_ducky #65: Character "X" not coming trough + wrong keyboard layout

careyer, 2017-08-31:
Hi,

I just got my ducky working... however I realized when printing "ABCDEFGHIJKLMNOPQRSTUVWXYZ" that the output sequence will only be: ABCDEFGHIJKLMNOPQRSTUVW

I tryied to print just the letters XYZ and it did not work...
printing the String "YZ" will yield in ZY (letters switched like in the EN keyboard layout).

Tried to print only "X" and this simply yields in no output at all.

What am I  getting wrong here?
Can anyone please give me a hint how to enable the right keyboard layout and how to get the "X" going?

Thanks!

tobozo, 2017-08-31:
could it be related to #31 ?

careyer, 2017-08-31:
@tobozo : Yes it seems to be kind of the same Problem... however in my case it is not the letter "P" that gets stuck but the letter "X" .... smal "x" however works fine.

I have tried different baud_rates in the arduino_wifi_duck.ino
define BAUD_RATE 57200   //"X" missing
define BAUD_RATE 57600   //"X" missing
define BAUD_RATE 115200   //does not work at all

I could solve the keyboard layout problem however! ;-)
Take a look at: https://github.com/BlueArduino20/LocaleKeyboard.BAT

careyer, 2017-09-06:
Can anyone reproduce this issue on their builds?
@spacehuhn: Stefan, from your youtube live broadcast together with Seytonic I know that you live near cologne - so do I =D. Would it be helpful to try debug this issue together if you can't reproduce it on your unit? Seems like there is a general problem as it seems to be related to #31 as well. Cheers & keep up the excellent work!

supersjimmie, 2017-09-07:
So you're saying that a code like this:
```
DELAY 500
STRING Testing 123
ENTER
DELAY 100
STRING X
ENTER
DELAY 100
STRING Test finished
ENTER
```
Will only print:
```
Testing 123
Test finished
```
If you can just confirm that, or post an exact reproducing ducky script, I will hook-up my wifiducky stick and my keylogger-ducky and test it for you.

careyer, 2017-09-07:
@supersjimmie : Exactly! The above test script exactly reproduces the issue.
moreover the char "X" will also be missing in the middle of a string:

`DELAY 500`
`STRING Testing ABCXYZ`
`ENTER`
`DELAY 500`
`STRING Test finished`
`DELAY 500`
`ENTER`

will produce the Output:
`Testing ABCYZ`
`Test finished`

careyer, 2017-09-11:
@supersjimmie : Just curious: could you reproduce the issue on your hardware?

supersjimmie, 2017-09-12:
@careyer sorry started at a new job, not enough free time at the moment.  :(
Will do asap!

supersjimmie, 2017-09-13:
@careyer I found some time.  :)
Script:
```
DELAY 500
STRING Testing ABCXYZ
ENTER
DELAY 500
STRING X
ENTER
DELAY 500
STRING XX
ENTER
DELAY 500
STRING Test finished
DELAY 500
ENTER
```
Result:
```
Testing ABCXYZ
X
XX
Test finished
```
So that concludes that there is something different between your setup and mine, that causes your problem.

To start with, I have made a combination of the hardware and the code from this Ducky and the Keylogger. But this mix should not make any difference to the ducky part.

My hardware (real keyboard and the pc) are setup for a US (International) keyboard.
If your layout is so different that your "X" is at another location, we cannot compare 1-to-1.
 
I use "Serial" for the debugging part at 115k2, and I use softwareserial (in my code defined as "ExternSerial") for the communication between the esp8266 and the Arduino at 57k6.  Please check at least that you don't have a slight difference between the baudrate in the esp code and the arduino code. 

Please try 57k6 (57600) because I did notice problems when I used the original ```#define BAUD_RATE 57200```. So I used ```#define BAUD_RATE 57600``` And again, do that in both the esp code and the arduino code.

Please keep us informed.

supersjimmie, 2017-09-13:
Ah and now I see that you mentioned a German layout?
So at the position of your X is a C here.
I changed my script to test it with a C (and than again with a Y), same good results.

careyer, 2017-09-15:
Hi @supersjimmie : Thank you for testing! Indeed I tweaked a bit in order to be able to use a german keyboard layout - however I also tried with the original US layout. The issue stays the same for both. (The letter X is at the same position in both layouts only Y  & Z are swapped).

I also tried playing around with the baudrate on both sides (Arduino / ESP) and changed them from 57200 to 57600. This also did not cause any change. I am a bit clueless what I can do more to debug and haunt this issue down?

supersjimmie, 2017-09-15:
I'm still suspecting the baudrate.  You have an issue with the 'X', while issue #31 is about the 'P'. Now 'X' is hex 58 bin 0101**0**000 and 'P' is hex 50 bin 0101**1**000. So they differ 1 bit. I cannot thing of a reason why, but it looks like something goes wrong there in the serial communication.

Can you try to lower the baudrates extremely to something like 9600?
Also, I once had a slower device instead of the arduino and then I had to put a small delay at the tx (esp) side:
```
#define TX_DELAY 2
void sendBuffer() {
  for (int i = 0; i < bc; i++) {
    mySerial.write((char)scriptBuffer[i]);
    delay(TX_DELAY);
  }
  runLine = false;
  bc = 0;
}
```
Perhaps you can try that too?

careyer, 2017-09-24:
@supersjimmie your guess was right: I reduced the baud rate on both sides (Teensy + ESP8266) to 19200 baud and everything works flawlessly now! Thanks for your help

Mr-iX, 2017-09-25:
Does for somebody the character "&" work?
When yes, which keyboard layout are you using?

careyer, 2017-09-28:
@Mr-iX : in german keyboard layout "&" doesn't work either

Mr-iX, 2017-09-29:
But for the "&", a lower baud rate doesn't help. It looks like there is something in the Keyboard.cpp wrong.