spacehuhn/wifi_ducky #67: STRING stops after 6 characters

AnthonyMentesana, 2017-09-16:
when i run a script it will run up to a certain point and then doesnt print the full string just the first 5 characters, i think this may be an issue with the serial communication
I have changed the Buad to 57600 on both, but it had to use the upload to get it to the ESP and im not sure if anything changes. it doesnt seem to work when uploading.
using an esp 12e and CJ MCU like the beetle with a 3.3v voltage regulator. tested with a NODE MCU and the same beetle and it had the same issue.
could it be the voltage regulator or the Baud rate?

AnthonyMentesana, 2017-09-16:
i think it has more to do with the number of lines in the script... after about 8 lines it has trouble

careyer, 2017-09-24:
Does the Script halt at the same position in the script all the time (disallowed/unsupported character?). If the problem differs from run to run you might have issues with an unstable serial connection. You should reduce serial speed then. See #65