spacehuhn/wifi_ducky #84: Seeking Clarification-Project Working Great!

B4UC2FAR, 2017-10-08:
This is the third of your projects I have built and would like to say thanks for sharing and for taking the time to support them. I am looking for clarification where you mention connecting to 192.168.4.1/update to load the .bin file. Is this .bin file you are referring to an actual ducky script being uploaded from another source besides the live execute window or is this .bin file you are referring to an actual update for the ESP8266 SDK such as was flashed onto it in the beginning steps? Thanks in advance for your time.

Mr-iX, 2017-10-08:
The update section is for flashing a new firmware to the ESP. 
When you want to upload a Ducky script, you can do that on the scripts webpage:
[image]

B4UC2FAR, 2017-10-08:
Thanks for the clarification. It’s handy that the firmware can be updated from the browser as additional features are added.

closed 2017-10-08 by B4UC2FAR