spacehuhn/wifi_ducky #86: Left out project

harshil2105, 2017-11-22:
I know this is a one man show and I'm really happy to see your progress in nodemcu wifi deauther. But I will request you to make some effort on wifi ducky project too. This is one of the best project,you just need to make the software better(web design,certain features etc) I hope you would work on this project too

harshil2105, 2017-11-22:
There are also lot of bugs in the initial software. Please help

careyer, 2018-03-29:
+1 .... DeAuther has been completely reworked an it would be great to see this for this awesome project as well! Some things like flashing new versions via the webinterface do not work in the current version.

harshil2105, 2018-03-29:
It's a laggy version!!!