spacehuhn/wifi_ducky #91: Has anyone been able to get the all in one boards from aliexpress to work?

samr28, 2018-05-17:
I purchased this board: https://www.aliexpress.com/item/ATMEGA32U4-ESP8266-ESP-12E-ESP12E-Badusb-TF-Micro-SD-Virtual-Keyboard-Development-Board-for-Arduino/32839674193.html?spm=a2g0s.9042311.0.0.R4hbkG

My computer picks it up as an arduino leonardo. So I uploaded the code that allows you to flash the esp board (https://gist.github.com/spacehuhn/b2b7d897550bc07b26da8464fa7f4b36.) Then I tried to use nodemcu to flash it but got the following errors:
```
Note:Detect serial port changed.
Note:Auto MAP serial port.Port-->COM7

Error:Serial port connect failed, please check it.
```

Has anyone else been able to get this board or one similar to it working? The bottom says CJMCU-3212 if that helps.

terrabitz, 2018-05-24:
There is just one tutorial that I have found for this, but it is entirely in Chinese:  https://www.bilibili.com/video/av22645681. If anyone understands it, I would love to figure this out as well. I've been trying to flash just the ESP8266 using a USB-to-TTL device, but haven't had any luck. According to the video, the 4 pins on the right-most side under the "CJMCU" label are  Rx, Tx, Vcc, and GND respectively. I've been connecting to those four and then connecting the two circles on the top left, which I presume is to pull CH_PD up. I've been staring at this board for hours, and still haven't been able to make much progress. I'll probably try and pull the board apart once I get some better soldering equipment.

wilson289296, 2018-05-24:
As luck would have it, I speak mandarin. I've watched the video and I think I have the steps down. I'll try to get my hands on that adapter he mentioned and I'll see if it works. In short, though (not having verified my understanding), you grab that adapter, match the holes on the 3212 board to the terminations on the adapter, plug both the badusb and the adapter into your computer, use arduino ide to flash it. But during the entire time you do so, you need to make sure you connect the two silver metallic circles on the upper right side of the TOP of board (this grounds the "firmware flash" pin on the 8266, telling the 8266 that it's supposed to be in fw flashing mode) and hold the white button on the board for the entire duration of the flash. He didn't mention what he used, but I'm assuming it's just the standard nodemcu flasher according to this project's documentation. See if you guys can work it out; if you need more translated details, let me know and I'll give line by line translation. Good luck to all!

pjaakko, 2018-06-03:
It's nice to find out, that I'm not only one trying to solve the mystery of this CJMCU-3212 badusb device :-)

Has anyone managed to figure out how to use the default firmware? When I powered the device, it creates WLAN AP with SSID "AI-THINKER_<hex>" without password. ESP assigns IP address from 192.168.4.0/24 network but there isn't http server or anything running.

samr28, 2018-06-04:
Mine is the same as @pjaakko. I think we need to figure out how to flash the ESP to get the http server running. From what I understand, to do this, we need to get the leonardo to run the flasher arduino sketch which basically just forwards everything to the ESP board. Then we can put the real arduino sketch on it and we should be good.

I also tried an IP scanner to see if it was just running on a different ip or something but I didn't find anything running so I think the http server is just not running.

Maybe @wilson289296 can re check the video to see what exactly we should be doing for flashing. I tried connecting the two metallic circles but when I press and hold the button (which appears to be a reset button), the device will not turn on. So maybe it was some sort of miscommunication or translation error? I'm not quite sure.

wilson289296, 2018-06-04:
I'll try to get a line-by-line translation of the video done by the end of this week, sorry for the slow progress.

SmartContractsAgency, 2018-06-12:
@wilson289296
thank you
we are waiting)

I hoped that we can flash esp without any additional adapters.

also I think that we can check number of pins connected to esp and atmega
program_pin and enable_pin

maybe it is not 12/13
can we check with multimeter?

pjaakko, 2018-06-29:
We seems to be more or less stuck here. Is there any other community, forum or such to discuss about this aliexpress device?

puckk, 2018-07-19:
@wilson289296 some news?

samr28, 2018-07-27:
I went through that video again and found some info on the pins: https://i.imgur.com/MgTCnhV.jpg. But other than that, no progress :(

EurophiaDealer, 2018-08-07:
Hello guys, im new at this tread. So do we have anything new?
@wilson289296 did you translated the video?

EurophiaDealer, 2018-08-07:
Iam trying to get contact with the video creator at bilibili.

EurophiaDealer, 2018-08-07:
For god sake!!!. I can't send a message to the bilibili guy cus i don't have level in the plataform.

EurophiaDealer, 2018-08-07:
Anyone, try to contact this guy, https://space.bilibili.com/39602674/#/, he is creator of the only vídeo in the world explaning how to configure the CJMCU-3212. I don't know how do i get this "trustness level" to contact someone at bilibili.

puckk, 2018-08-07:
Hi, I was able to make it work! I made a guide for you

https://github.com/puckk/CJMCU-3212

EurophiaDealer, 2018-08-07:
i lv you man

AieCaPique, 2018-08-20:
Hello 
 Thank you puckk for your tutorial :)
It works great for me with the flash version for windows
Can you get the micro SD slot to work?

puckk, 2018-08-20:
Yes, the SD card worked with the examples of the Arduino IDE.

AieCaPique, 2018-08-24:
Puckk could you add to your tutorial the method to operate the SD slot ?
thank you very much

FietsenRijder, 2018-08-27:
Hello,
I'm planning on buying one of these. Would you recommend them? And @Puckk thanks already for your tutorial, but is there a way on looping the two circles without that special wire or can you just do it with something else?
Thank you

samr28, 2018-08-27:
@FietsenRijder You should just need to short those two pads together. Any object that conducts electricity (like a paperclip) should work.

FietsenRijder, 2018-08-28:
@samr28 Thanks for your quick respond do you recommend it and does it work?

AieCaPique, 2018-08-29:
yes for me it works very well
I just need to make conf for the micro SD is supported

FietsenRijder, 2018-09-12:
@puckk So the thing is delivered but I don't understand how to flash it I did step 1 of your tutorial and when i plug the usb in a red light on top turns on but I don't understand step 2. First how to install esptool? And what to do then i'm new so can you be a little more specific?
Thank You

samr28, 2018-09-12:
@FietsenRijder What OS is your computer? You need to install python and then follow the instructions here: https://github.com/espressif/esptool#installation--dependencies

FietsenRijder, 2018-09-17:
@samr28 I'm running the latest windows 10. I do not understand how to install. @AieCaPique  @puckk  What sd card are you using and does te 6 pin thing that comes with it work to bridge?
And @puckk  do you have to bridge first and then plug in or how did you do that, I do not understand it at all.

Please help me

FietsenRijder

riogesulgon, 2018-09-24:
Hi,
Please help to troubleshoot the steps I am using. There seems to be something I am missing.
1. I have uploaded step1.ino using the Leonardo board. It seems that the program is setting the following pins:

- ENABLE to low 
- PROGRAM to high

2. Removed the device from usb socket and soldered wires to the top two right pads so i can easily join and remove the ends of the wires.
3. Inserted the device into the usb port and ran the command.
However I am getting the following error message:

esptool.py.exe --trace   --baud 115200 --port COM9 write_flash 0x00000 /dev/esp8266_wifi_duck_4mb.bin --flash_size 4MB --flash_mode dio --flash_freq 40m
esptool.py v2.5.0
Serial port COM9
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.

[image]

I am running this on Windows 10 by the way and will try to run on Mac OS later as the interaction between serial port and esptool might not be working properly. Appreciate any help you might provide. Thanks.

samr28, 2018-09-24:
@riogesulgon I had the same issue with esptool. To fix it, I used nodemcu which worked the first try.

riogesulgon, 2018-09-25:
Thanks@samr28! I was able to flash the firmware using the nodemcu flasher (https://github.com/nodemcu/nodemcu-flasher). My board is now working properly.

AieCaPique, 2018-11-28:
Hello everyone
can you get the card to work with the Micro SD slot?

I thank you in advance for your help because me on my side the SSID works but impossible to operate the Micro SD

chepo92, 2018-12-27:
@AieCaPique the SD slot seems to be wired to the Atmega, at least the example sketchs are working for me (use SS pin 8)

TheIronShadow, 2019-01-18:
I have a question. so if you follow the guide the wifi mode and sd card mode will work?. don't you have to upload the duckduino-microsd or duckduino-microsd/dip.

and how to make the wifi and microsd with dip switch working?

AieCaPique, 2019-02-28:
@TheIronShadow  for me wifi works, code execution also works.
The only thing that does not work is the micro sd part to store more code or executables as would a rubber ducky

cristyanul, 2019-03-03:
@AieCaPique if you get it to work with the SD card please create a clear tutorial on everything.

cristyanul, 2019-03-03:
If someone could take the code from http://seytonic.com/2016/11/26/arduino-rubber-ducky-microsd/  and merge it with the wifi code it would be great.

kariez, 2019-04-01:
@puckk did you get the SD card working somehow? Trying it atm. It would be awesome if you could share the code.

kariez, 2019-04-01:
You could use the SD card to pull and run files from it. Paxy's duckuino has this kind of capability. Source to paxy's code, links in desc. -> https://youtu.be/3A53g0j-ugw 
He has not made it for the WiFi version tho as you can see, but it should work with little bit of tweaking. Remember to share if you get it working.

rl34075, 2019-04-05:
Please, look at www.l2r1.com.br. There are some scripts to configure your cjmcu esp8266.

AieCaPique, 2019-04-18:
Thanks @rl34075  for the tutorial
on my side I am step by step the procedure but the script.txt does not work
more since I followed the indication I can not write the characters & and% with the function ALT NUM_3 NUM_7 for example for the%

If someone has an idea

rl34075, 2019-04-18:
I think the problem is in keyboard layout... The script is to english
Layout... maybe you need to remap ekyboard keys.

TechieRuss, 2019-05-05:
I think I solved the mystery of not being able to flash the esp8266 using the atmega32u4 as a serial pass-through. 
A banggood seller posted this "schematic" [http://myosuploads3.banggood.com/products/20190102/20190102202550CJMCU-32U4TFWIFIforreference.pdf](http://myosuploads3.banggood.com/products/20190102/20190102202550CJMCU-32U4TFWIFIforreference.pdf) 
As you can see this thing has so many errors that if any of the Chinese fab houses are actually following it I'm shocked that these cards work at all.

I was able to flash the ESP by de-soldering it off of the assembly and using an arduino uno as serial passthrough. 

And now I'm off to KiCad to make my own board that will work.

camilitox, 2019-05-15:
@puckk Your tutorial was amazing, do you have some script to use de SD card simultaneously with the WiFi?

rl34075, 2019-05-15:
I'm working on it.... I know that is possible to use SD as a server file
for ESP8266...so, you can upload files to the server (sd card in this
case). You could exfiltrate data from target host to Server uploading to SD
or use SD card as a exlpoit repository to use with server...

*Leonardo La Rosa*

TechieRuss, 2019-05-15:
On the board that I got, the SD card is wired to the atmega32u4 not to the esp8266. this might make it more difficult to move files to it from the 8266.

rl34075, 2019-05-15:
I think that you can use arduino to connect everything together.

*Leonardo La Rosa*

camilitox, 2019-05-25:
https://github.com/joelsernamoreno/badusb_sd_wifi?files=1 ... This can read scripts from sd ... But uses other language script

blackecho101, 2019-06-05:
We need to find a way to merge the SD capability of badusb_sd_wifi with the features and ducky script compatibility of this amazing project! I ordered my board from aliexpress and it arrived in 12 days! I hope there is more activity around this board soon!

kryptonut, 2019-07-29:
It's a great little board.  Question, does it just read raw duckyscript in the interpreter, or does it still need an inject.bin of sorts?  Might see if I can cobble something together to get the SD card working too - no promises though, I'm not an actual developer.

rl34075, 2019-07-29:
Hi Glen,

You have to flash firmware to configure the board. After configure, you
only need to upload the scripts from Web or to SD card.

*Leonardo La Rosa*

rl34075, 2019-07-29:
I think it's possible... You can use SD card as a script library...

*Leonardo La Rosa*

rl34075, 2019-07-29:
Hi AieCaPique,

I've changed the scripts to use pt-BR keyboards... you can google for
keyboard mapping in arduino and change it to satsfy your needs.

*Leonardo La Rosa*

kryptonut, 2019-07-29:
Mine is flashed, and I can upload scripts, run live scripts, etc.

I've had no success putting scripts on the SD card and from what I've seen in the code, there's no reference to any sort of SD libraries or SD card handling.

rl34075, 2019-07-30:
Hello Glen,

Take a look at this links...I think they can help you:

https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/examples/SDWebServer/SDWebServer.ino
https://www.hackster.io/draw2dtouch/nodemcu-webserver-with-sd-card-support-169adc

*Leonardo La Rosa*

code2319, 2019-09-25:
> Hello Glen, Take a look at this links...I think they can help you: https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/examples/SDWebServer/SDWebServer.ino [...]

can i clarify: i need to combine SDWebServer.ino with esp8266_wifi_duck.ino ?

PavKmiec, 2019-10-14:
Does anyone have a problem of not being able to connect to the web interface? 
I used the puckk's instructions before and it worked like a charm.

I got a new device yesterday, flashed it, uploaded the sketches and it did work until I changed something in settings. Since then It won't connect to web interface anymore, keeps loading then "connection refused" and I could use some help. 

What I did was to go an do all the uploading / flashing which works fine again but it just won't connect to the interface.

Does anyone know how to fix that? I's there a way to completely wipe it or is my device defective? 

thanks

vtontodonato-c, 2019-10-24:
I need to flash esp8266 on the CJMCU-3212 via OTA or access the esp8266 via ARDUINO IDE to develop my own software NOT duck.

Is anybody trying something as this?

Can I flash my arduino code directly?

thanks

chepo92, 2019-11-20:
As I said before the SD is wired to the Atmega, so you upload scripts to esp8266 then send them via serial to Atmega then it has to write to sd

neoskyriper, 2019-12-03:
Pour ma part je n'arrive plus a le flasher avec un .bin avec nodemcu-flasher-master ESP8266Flasher.exe
je suis désespéré je ne sais plus quoi faire 
si vous avez une solution s'il vous plait merci

sebastienPoussard, 2019-12-22:
adter a few months, i suceeded to make it work, it seems I needed to to it with Windows.
With my Linux laptop it didn't worked.
Anyway, I'm now facing a new issue, when i type text in the web server, it is mappy to different code on the remote host (azerty) @AieCaPique any ideas ?

cnu77, 2020-01-30:
> Hi, I was able to make it work! I made a guide for you
> 
> https://github.com/puckk/CJMCU-3212

I tried, but not working.

I tried 3 different ways, ending in the same loop. Tried joining the 2 pins as shown in your pic, same issues.

1. esptool.py --> serial.serialutil.SerialTimeoutException: Write timeout
2. Arduino IDE --> serial.serialutil.SerialTimeoutException: Write timeout the selected serial port serial.serialutil.SerialTimeoutException: Write timeout does not exist or your board is not connected
3. NodeMCU Flash --> Waiting MAC

I am not sure, if...

1. GPIO0 is grounded or not (Or, by default, it is grounded)
2. Do I need to flash ESP8266 with firmware? Is it a 'must'?

Any suggestions, please?

Xentraxx, 2020-02-06:
I got WiFiDucky working but as I'd like to use different keyboard mappings I'd like to use the successor WiFiDuck as it supports different keyboard types. 
@rl34075 You mentioned that you changed the script to use the pt-BR keyboard. Could you explain or provide a resource on how to use a qwertz keyboard with WiFiDucky for example?

[11 more replies]