spacehuhn/wifi_ducky #105: Erro ao conectar na rede

Nikakz, 2020-07-28:
Olá, sou noob, executei e tudo correu bem, mas quando tento conectar na rede wifi duck, usando win10 PRO ele me retorna dizendo que não foi possível conectar-se a essa rede. Alguém poderia ajudar? Não consegui achar solução para meu problema. Obrigada

jamezrin, 2020-07-28:
Please use english my fellow human being

Nikakz, 2020-07-29:
Hello, I'm noob, I ran and everything went well, but when I try to connect to the wifi duck network, using win10 PRO he returns to me saying that it was not possible to connect to that network. Could anyone help? I couldn't find a solution to my problem. Thanks

Nikakz, 2020-07-29:
> Please use english my fellow human being

Hello, I'm noob, I ran and everything went well, but when I try to connect to the wifi duck network, using win10 PRO he returns to me saying that it was not possible to connect to that network. Could anyone help? I couldn't find a solution to my problem. Thanks